khoih-prog/EthernetWebServer_SSL #1: Problem with ESP32 (Wemos D1 R32) SPI?

1simone0, 2021-01-28:
I have a Wemos D1 R32 (basically an ESP32 with UNO layout) and I'm trying to make it work with an Original Arduino Ethernet Shield 2.
Here a link with a Wemos D1 R32  pinout description: [https://www.botnroll.com/en/esp/3639-wemos-d1-r32-w-esp32-uno-r3-pinout.html](url)

I started from a simple example:  "**simpleHTTPExample**" but with no good results.
I checked the pin used from the library in case of ESP32 (18-19-23-5) that  are ok with that shield but I don't see any signal with scope on them. Not sure about SS pin really used (5 or 22) but reading the serial monitor it seems it takes pin 5 at the end. Anyway for sure on 18-19-23 there are no signals. 

From original sketch I modified first row of MAC adress in defines.h. Also tried to change the ethernet library managing trues and falses in define.h file.  Am I missing something more to manage? 

Here my Serial monitor:

Starting SimpleHTTPExample on ESP32 with W5x00 using Ethernet Library
EthernetWebServer_SSL v1.3.1
[ETHERNET_WEBSERVER] =========== USE_ETHERNET ===========
[ETHERNET_WEBSERVER] Default SPI pinout:
[ETHERNET_WEBSERVER] MOSI: 23
[ETHERNET_WEBSERVER] MISO: 19
[ETHERNET_WEBSERVER] SCK: 18
[ETHERNET_WEBSERVER] SS: 5
[ETHERNET_WEBSERVER] =========================
[ETHERNET_WEBSERVER] ESP32 setCsPin: 22

Currently Used SPI pinout:
MOSI:23
MISO:19
SCK:18
SS:5

Using mac index = 17
Connected! IP address: 0.0.0.0
Connect failed: -1

Any suggestions?

khoih-prog, 2021-01-28:
The Arduino Ethernet Shield is designed for Arduino (Mega, UNO, SAM DUE) boards, and using ICSP connector for SPI signals (MOSI, MISO, SCK, CS).

[image]

[image]

You have to connect the SPI signal from ICSP connector into corresponding pins to use it.

Also remember the GPIO 23, 19, 18, 5 (MOSI, MISO, SCK, CS) are already connected to pins 11,12,13, 10 of the connector.

closed 2021-01-28 by khoih-prog

1simone0, 2021-01-28:
In fact, I'm saying that from those pins (23,19,18,5) there are no signals!!! It seems ESP32 is not well driven. I'm wondering if someone else used this library with an ESP32 chip first and with wich kind of board to figure out if I'm making a mistake or missing something or if it's a library problem.

khoih-prog, 2021-01-28:
I just used the similar to Wemos D1 ESP32 + W5500 yesterday with the twin library [**EthernetWebServer**](https://github.com/khoih-prog/EthernetWebServer) and the combination are working perfectly. The Ethernet shield is W5500, not the Arduino W5500 Ethernet shield 2, but they are very similar. You have to connect the wire manually. That's it.

[image]

Have a look at the result:

[**ESP32_FS_EthernetWebServer on ESP32 using LittleFS with W5x00 using Ethernet Library**](https://github.com/khoih-prog/EthernetWebServer#5-esp32_fs_ethernetwebserver-on-esp32-using-littlefs-with-w5x00-using-ethernet-library)