khoih-prog/EthernetWebServer_SSL #2: EthernetSSLClient cannot be passed as an argument to a Class

RoSchmi, 2021-02-15:
I'm not sure if this is an issue or a lack of my programming skills but I'm busy for many hours on this and can't get it solved.

I'm working on a Wio Terminal board using PlatformIO IDE. I'm working with an ENC28 module on the Wio Terminal.
I followed your example EthernetWebServer_SSL - 'WebClient_SSL' and the example worked as expected.
In an application I'm working on, I used the Seeed rpcWiFi library and successfully passed the created WiFiClient and WiFiClientSecure objects to another Class and called methods of this class from the Arduino loop.
Now I wanted to do the same using an Enc28 Ethernet module but I could not get it working.
If I pass the EthernetSSLClient instance to the Class in which i want to use it and make an '#include "EthernetWebServer_SSL.h"' statement in this class my application doesn't build because of multiple definition of 'EthernetWebServer' methods.
I made a short code sample which was just uploaded to GitHub.
-https://github.com/RoSchmi/Wio_Terminal_Enc28_Ethernet_Https_Test

It would be nice if I could use the Ethernet based Client as it works with the WiFiClient.

khoih-prog, 2021-02-15:
You can do either of the following ways after a good rest then a cup of coffee

### 1. Using all h files
Just changing these 2 files

#### 1.1 MyClass.h

```
#ifndef _MYCLASS_H_
#define _MYCLASS_H_

class MyClass
{
  public:

    MyClass(EthernetSSLClient * pClient);

    void DoSomething();

  private:
    EthernetSSLClient * myEthernetSSLClient = NULL;

};

#include "MyClass_Impl.h"

#endif
```

#### 1.2 MyClass.cpp => MyClass_Impl.h

```
#ifndef _MYCLASS_IMPL_H_
#define _MYCLASS_IMPL_H_

   
MyClass::MyClass(EthernetSSLClient * pClient)
{ 
   myEthernetSSLClient = pClient;
}

void MyClass::DoSomething()
{     
    myEthernetSSLClient->connect("www.google.de", 443);
}

#endif    // _MYCLASS_IMPL_H_
```

### 2. Using only MyClass.h file

#### 2.1 MyClass.h

```
class MyClass
{
  public:

    MyClass(EthernetSSLClient * pClient)
    {
      myEthernetSSLClient = pClient;
    }

    void DoSomething()
    {
      myEthernetSSLClient->connect("www.google.de", 443);
    }

  private:
    EthernetSSLClient * myEthernetSSLClient = NULL;

};
```

#### 2.2 delete MyClass.cpp

closed 2021-02-15 by khoih-prog

RoSchmi, 2021-02-16:
Thank you very much for your help. I think that I wouldn't have found the solution by myself. I didn't yet use and understand the concept of having the implementation in header files.
I hope that you don't find it too much inappropriate to post this programming problem in the 'Issues' section.
So, after following your advice, I learned that apparently my code didn't build because of a violation of 'One Definition Rule'. Using the WiFiClient presumably my concept worked, as the passed object was created with a parameterless constructor whereas the EthernetSSLClient instance can only be created with a parameterized constructor.

Btw: May I ask which 'driving force' is behind your repository, which produces so much valuable code to enable useful libraries to be used on so many different boards? It's unbelievable that this is done by one person.

khoih-prog, 2021-09-04:
Hi @RoSchmi 

Could you please try again using [**QNEthernet** library](https://github.com/ssilverman/QNEthernet) ?

In case you haven't tried it, have a quick look at this very good [**New lwIP-based Ethernet library for Teensy 4.1**](https://forum.pjrc.com/threads/68066-New-lwIP-based-Ethernet-library-for-Teensy-4-1)

---

### Major Releases v1.6.0

1. Add support to [QNEthernet Library](https://github.com/ssilverman/QNEthernet) for Teensy 4.1 built-in Ethernet
2. Update examples with new features

RoSchmi, 2021-09-06:
Hi @khoih-prog,
I think you mean this issue:
https://github.com/khoih-prog/EthernetWebServer/issues/31
This question was concerning Wio Terminal and ENC28 module.

khoih-prog, 2021-09-06:
Oh, sorry ;={