khoih-prog/EthernetWebServer_SSL #3: File 'Ethernet_HttpClient.cpp' doesn't compile for Teensy 4.1 on PlatformIO

RoSchmiSec, 2021-03-09:
When compiling 'Ethernet_HttpClient.cpp' I get the following error message:

In file included from C:\Users\Roland\.platformio\packages\framework-arduinoteensy\cores\teensy4/HardwareSerial.h:106:0,
                 from C:\Users\Roland\.platformio\packages\framework-arduinoteensy\cores\teensy4/WProgram.h:46,
                 from C:\Users\Roland\.platformio\packages\framework-arduinoteensy\cores\teensy4/Arduino.h:6,
                 from .pio\libdeps\teensy41\EthernetWebServer_SSL\src/Ethernet_HTTPClient/Ethernet_HttpClient.h:31,
                 from .pio\libdeps\teensy41\EthernetWebServer_SSL\src\Ethernet_HTTPClient\Ethernet_HttpClient.cpp:32:
C:\Users\Roland\.platformio\packages\framework-arduinoteensy\cores\teensy4/Stream.h:68:6: error: 'int Stream::timedRead()' is private        
  int timedRead();

This is the causing code segment:

    // KH test
    int c = timedRead();
    //int c = iClient->read();

When I change to:

    // KH test
    //int c = timedRead();
    int c = iClient->read();

it compiles.
Can this be done without risk?

khoih-prog, 2021-03-10:
Hi @RoSchmiSec 

Thanks again to use the library.

I forgot to add in the README that you have to copy [Stream.h](https://github.com/khoih-prog/EthernetWebServer_SSL/blob/main/Packages_Patches/hardware/teensy/avr/cores/teensy4/Stream.h) to replace the original Stream.h located at  `./arduino-1.8.xx/hardware/teensy/avr/cores/teensy4/Stream.h`

Have a look at

[Packages' Patches For Teensy boards](https://github.com/khoih-prog/Ethernet_Manager#2-for-teensy-boards)

> 
> #### 2. For Teensy boards
>  
>  **To be able to compile and run on Teensy boards**, you have to copy the files in [**Packages_Patches for Teensy directory**](Packages_Patches/hardware/teensy/avr) into Teensy hardware directory (./arduino-1.8.13/hardware/teensy/avr/boards.txt). 
> 
> Supposing the Arduino version is 1.8.13. These files must be copied into the directory:
> 
> - `./arduino-1.8.13/hardware/teensy/avr/boards.txt`
> - `./arduino-1.8.13/hardware/teensy/avr/cores/teensy/Stream.h`
> - `./arduino-1.8.13/hardware/teensy/avr/cores/teensy3/Stream.h`
> - `./arduino-1.8.13/hardware/teensy/avr/cores/teensy4/Stream.h`
> 
> Whenever a new version is installed, remember to copy this file into the new version directory. For example, new version is x.yy.zz
> These files must be copied into the directory:
> 
> - `./arduino-x.yy.zz/hardware/teensy/avr/boards.txt`
> - `./arduino-x.yy.zz/hardware/teensy/avr/cores/teensy/Stream.h`
> - `./arduino-x.yy.zz/hardware/teensy/avr/cores/teensy3/Stream.h`
> - `./arduino-x.yy.zz/hardware/teensy/avr/cores/teensy4/Stream.h`

Will update README shortly.

Regards,

closed 2021-03-10 by khoih-prog

RoSchmi, 2021-03-10:
Many thanks for your help,
I have to apologize that I didn't read the README in detail, so I was not aware that you already had a fix for this issue.
I assume, that on PlatformIO the right way is to replace the Stream.h file in  C:/Users/thisUser/.platformio/packages/framework-arduinoteensy/cores/teensy4/Stream.h respectively.
 I searched some time for a way to overwrite Stream.h with a custom version through an entry in platformio.ini but I was not successful.
Btw: I'm happy to confirm that EthernetHttpClient_SSL works reliable (in short term tests) using an ENC28 module with EthernetENC library on the Wio Terminal board.
Actually I'm on the way to port my application on Teensy 4.1.

Many thanks again. You should definitely have a 'Donation.Button' on your page.