khoih-prog/EthernetWebServer_SSL #6: Buffer limitation?

gitTinker, 2021-05-29:
When I run this PoC code the browser console states:
**net::ERR_CONTENT_LENGTH_MISMATCH 200 (OK)**

If just one character is removed, no error occurs.

The size of the variable containing the contents is 2050.
I have set #define SENDCONTENT_P_BUFFER_SZ 6144 (event though the default 4096 should be sufficient)

The PoC attached here is merely the example, with the handelRoot function replaced, the inclusion of a file, and these lines:

```
#include "index_html.h"
#define SENDCONTENT_P_BUFFER_SZ 6144

void handleRoot() {
  Serial.print(F("SENDCONTENT_P_BUFFER_SZ: ")); Serial.println(SENDCONTENT_P_BUFFER_SZ);
  Serial.print(F("INDEX_HTML: ")); Serial.println(sizeof(INDEX_HTML));
  server.send(200, "text/html", INDEX_HTML);
}

```
(remove the .txt extension to use the files)

[AdvancedWebServer_NativeEthernet_BufferError.ino.txt](https://github.com/khoih-prog/EthernetWebServer_SSL/files/6565426/AdvancedWebServer_NativeEthernet_BufferError.ino.txt)
[defines.h.txt](https://github.com/khoih-prog/EthernetWebServer_SSL/files/6565427/defines.h.txt)
[index_html.h.txt](https://github.com/khoih-prog/EthernetWebServer_SSL/files/6565428/index_html.h.txt)

gitTinker, 2021-05-29:
I am using a Featherwing M4 (SAMD51) with the [NIC](https://www.amazon.com/gp/product/B08KTVD7BR) (W5x00)

EthernetWebServer_SSL v1.5.0
Arduino IDE 1.8.15

khoih-prog, 2021-05-30:
I tested the code and see **no issue even with buffer size larger = 2075 bytes**

[image]

I suggest if you have more issue, spend some more time to **test, research** before posting something too early.
A good programmer is expected at least to ( code <-> test <-> investigate <-> research ) before surrender so soon.

Moreover, this library is just a **wrapper** on top of the Ethernet (Ethernet, EthernetLarge, Ethernet2, Ethernet3, etc) libraries and inherits all the related limitation. And many times the buffer size as well as handling can cause some issues.

It's also better that you start over and test using directly the Ethernet libraries, and see if there is any issue caused by them.

Unless you prove the bug is in this library, I'm closing the issue now.

closed 2021-05-30 by khoih-prog

gitTinker, 2021-05-30:
Am I using the buffer size correctly?
Is this a buffer issue?

gitTinker, 2021-05-30:
Thank you for investigating; every new bit of info helps.
Even the stereo-typical response of 'it works on my end' eliminates some variables.

Least you think I am an ignoramus, I am attaching a screenshot of the message in my browser.
[image]

gitTinker, 2021-05-30:
Thanks also for the suggestion to check [the root library](https://github.com/khoih-prog/EthernetWebServer); this behavior occurs there too.
You are right, your updates did NOT introduce this behavior.

gitTinker, 2021-06-06:
I found a work-around that lets me continue to use this very excellent library for most content, with a workaround for really big pages.

@khoih-prog, I think the limitation may be with the hardware.  I do not have any limitation with an ESP8266, but with [this shield](https://www.amazon.com/gp/product/B08KTVD7BR), on [this M4](https://www.adafruit.com/product/3857), I can only send 2050 characters - no matter what the SENDCONTENT_P_BUFFER_SZ is set to.
The key part of my functional example is sending in smaller chunks

```
  while (chunkStart < strlen(IMAGE_SVG)) {
    strncpy ( chunk, &IMAGE_SVG[chunkStart], maxSendBufferSize );
    chunk[maxSendBufferSize] = '\0';  // ensure null terminated
    client.print(chunk);
    chunkStart += maxSendBufferSize;
```
[image]

@khoih-prog, doesn't look like there is anything you can 'fix', so no expectation to re-open this

[AdvancedWebServer_NativeEthernet_BufferWorkAround.ino.txt](https://github.com/khoih-prog/EthernetWebServer_SSL/files/6605049/AdvancedWebServer_NativeEthernet_BufferWorkAround.ino.txt)
[defines.h.txt](https://github.com/khoih-prog/EthernetWebServer_SSL/files/6605050/defines.h.txt)
[index_html.h.txt](https://github.com/khoih-prog/EthernetWebServer_SSL/files/6605051/index_html.h.txt)

khoih-prog, 2021-06-07:
It's good that you found the temporary way out. 

Did you try to increase the buffer by follow [**For application requiring 2K+ HTML page**](https://github.com/khoih-prog/EthernetWebServer_SSL#1-for-application-requiring-2k-html-page) ?

gitTinker, 2021-06-07:
😲 again I learn something new!  I've read through the readme many times, but the relevance of that section just didn't occur to me...<sigh>.

However, the example I used is a 5K page and I like having a few simultaneous connections.  Reading [your notes ](https://github.com/khoih-prog/EthernetWebServer_SSL/blob/main/LibraryPatches/Ethernet/src/Ethernet.h) I could likely get up to 6K by dropping to a single socket but I am also running your [websockets ](https://github.com/khoih-prog/Websockets2_Generic) for automation and I think I'll leave as-is for this project.

I've left myself a note in my build instructions to try your Ethernet modifications for my next proj.

khoih-prog, 2021-06-07:
- I've read through the readme many times, but the relevance of that section just didn't occur to me....

I used to have that bad habit when still very young. But as time goes by, I've realized it's much more efficient and not wasting time by `not reinventing everything`. Then I've always tried to learn from other people's experience / works by reading carefully the docs / books and use / apply as much as I could.