khoih-prog/EthernetWebServer_SSL #7: Teensy 4.1 refuses to connect over SSL

gloveboxes, 2021-06-04:
Hi there, having problems getting the basic WebClient_SSL working.

On Ubuntu 20.04 LTS (AMD 64 bit) installed the

- Arduino 1.8.13
- TeensyArduino installer
- From Arduino library manager installed this EthernetWebServer_SSL library v 1.5.0
- Installed the patches for board and Teensy 4.1

I tried two samples:

The WebClient example from Arduino/libraries/EthernetWebServer_SSL/examples directory. It works, connects and downloads the Arduino ASCII art.

Starting WebClient on TEENSY 4.1 with Custom Ethernet using Teensy 4.1 NativeEthernet Library
EthernetWebServer_SSL v1.5.0
[ETHERNET_WEBSERVER] ======== USE_NATIVE_ETHERNET ========
[ETHERNET_WEBSERVER] Default SPI pinout:
[ETHERNET_WEBSERVER] MOSI: 11
[ETHERNET_WEBSERVER] MISO: 12
[ETHERNET_WEBSERVER] SCK: 13
[ETHERNET_WEBSERVER] SS: 10
[ETHERNET_WEBSERVER] =========================
[ETHERNET_WEBSERVER] Board : TEENSY 4.1 , setCsPin: 10
=========================
Currently Used SPI pinout:
MOSI:11
MISO:12
SCK:13
SS:10
=========================
Using mac index = 15
Connected! IP address: 192.168.1.119

Starting connection to server...
Connected to server
HTTP/1.1 200 OK
Server: nginx/1.4.2
Date: Fri, 04 Jun 2021 05:39:55 GMT
Content-Type: text/plain
Content-Length: 2263
Last-Modified: Wed, 02 Oct 2013 13:46:47 GMT
Connection: close
Vary: Accept-Encoding
ETag: "524c23c7-8d7"
Accept-Ranges: bytes

           `:;;;,`                      .:;;:.           
        .;;;;;;;;;;;`                :;;;;;;;;;;:     TM 
      `;;;;;;;;;;;;;;;`            :;;;;;;;;;;;;;;;      
     :;;;;;;;;;;;;;;;;;;         `;;;;;;;;;;;;;;;;;;     
    ;;;;;;;;;;;;;;;;;;;;;       .;;;;;;;;;;;;;;;;;;;;    

I tried the WebClient_SSL example from Arduino/libraries/EthernetWebServer_SSL/examples directory.

It builds and deploys but it never connects - I get the following information from the Arduino Serial Monitor and it doesn't appear to connect and does not return the Arduino ASCII art.

Start WebClient_SSL on TEENSY 4.1 with Custom Ethernet using Teensy 4.1 NativeEthernet Library
EthernetWebServer_SSL v1.5.0
[ETHERNET_WEBSERVER] ======== USE_NATIVE_ETHERNET ========
[ETHERNET_WEBSERVER] Default SPI pinout:
[ETHERNET_WEBSERVER] MOSI: 11
[ETHERNET_WEBSERVER] MISO: 12
[ETHERNET_WEBSERVER] SCK: 13
[ETHERNET_WEBSERVER] SS: 10
[ETHERNET_WEBSERVER] =========================
[ETHERNET_WEBSERVER] Board : TEENSY 4.1 , setCsPin: 10
=========================
Currently Used SPI pinout:
MOSI:11
MISO:12
SCK:13
SS:10
=========================
Using mac index = 6
Connected! IP address: 192.168.1.164
Connecting to : www.arduino.cc, port : 443

Do you have any idea why it is not working?

Thanks, Dave

khoih-prog, 2021-06-04:
The [**WebClient_SSL** example](https://github.com/khoih-prog/EthernetWebServer_SSL/tree/main/examples/WebClient_SSL) is working OK with many other boards.

It's possibly the NativeEthernet is not fully compatible with other Ethernet libraries as well as SSL layer.

This is the terminal output when running **WebClient_SSL on MBED RASPBERRY_PI_PICO with W5x00 using Ethernet Library** => OK

```
Start WebClient_SSL on MBED RASPBERRY_PI_PICO with W5x00 using Ethernet Library
EthernetWebServer_SSL v1.5.0
[ETHERNET_WEBSERVER] =========== USE_ETHERNET ===========
[ETHERNET_WEBSERVER] Default SPI pinout:
[ETHERNET_WEBSERVER] MOSI: 3
[ETHERNET_WEBSERVER] MISO: 4
[ETHERNET_WEBSERVER] SCK: 2
[ETHERNET_WEBSERVER] SS: 5
[ETHERNET_WEBSERVER] =========================
[ETHERNET_WEBSERVER] RPIPICO setCsPin: 5
=========================
Currently Used SPI pinout:
MOSI:3
MISO:4
SCK:2
SS:5
=========================
Using mac index = 14
Connected! IP address: 192.168.2.97
Connecting to : www.arduino.cc, port : 443
Connected to 104.18.29.45
Took: 1310
Using micros()
F0 50 A0 60 73 64 4A C2 2 48 89 AC 13 5E 18 A6 
Using analogRead()
20 35 48 5A 64 72 7F 86 91 9A 9F A5 AC B1 B6 BA 
HTTP/1.1 200 OK
Date: Fri, 04 Jun 2021 05:57:36 GMT
Content-Type: text/plain
Transfer-Encoding: chunked
Connection: close
Last-Modified: Wed, 02 Oct 2013 13:46:47 GMT
Vary: Accept-Encoding
Strict-Transport-Security: max-age=500; includeSubDomains
X-Content-Type-Options: nosniff
X-XSS-Protection: 1; mode=block
CF-Cache-Status: DYNAMIC
cf-request-id: 0a7732475d0000f99d10975000000001
Expect-CT: max-age=604800, report-uri="https://report-uri.cloudflare.com/cdn-cgi/beacon/expect-ct"
Server: cloudflare
CF-RAY: 659eecb89953f99d-YYZ
alt-svc: h3-27=":443"; ma=86400, h3-28=":443"; ma=86400, h3-29=":443"; ma=86400, h3=":443"; ma=86400

8d7

           `:;;;,`                      .:;;:.           
        .;;;;;;;;;;;`                :;;;;;;;;;;:     TM 
      `;;;;;;;;;;;;;;;`            :;;;;;;;;;;;;;;;      
     :;;;;;;;;;;;;;;;;;;         `;;;;;;;;;;;;;;;;;;     
    ;;;;;;;;;;;;;;;;;;;;;       .;;;;;;;;;;;;;;;;;;;;    
   ;;;;;;;;:`   `;;;;;;;;;     ,;;;;;;;;.`   .;;;;;;;;   
  .;;;;;;,         :;;;;;;;   .;;;;;;;          ;;;;;;;  
  ;;;;;;             ;;;;;;;  ;;;;;;,            ;;;;;;. 
 ,;;;;;               ;;;;;;.;;;;;;`              ;;;;;; 
 ;;;;;.                ;;;;;;;;;;;`      ```       ;;;;;`
 ;;;;;                  ;;;;;;;;;,       ;;;       .;;;;;
`;;;;:                  `;;;;;;;;        ;;;        ;;;;;
,;;;;`    `,,,,,,,,      ;;;;;;;      .,,;;;,,,     ;;;;;
:;;;;`    .;;;;;;;;       ;;;;;,      :;;;;;;;;     ;;;;;
:;;;;`    .;;;;;;;;      `;;;;;;      :;;;;;;;;     ;;;;;
.;;;;.                   ;;;;;;;.        ;;;        ;;;;;
 ;;;;;                  ;;;;;;;;;        ;;;        ;;;;;
 ;;;;;                 .;;;;;;;;;;       ;;;       ;;;;;,
 ;;;;;;               `;;;;;;;;;;;;                ;;;;; 
 `;;;;;,             .;;;;;; ;;;;;;;              ;;;;;; 
  ;;;;;;:           :;;;;;;.  ;;;;;;;            ;;;;;;  
   ;;;;;;;`       .;;;;;;;,    ;;;;;;;;        ;;;;;;;:  
    ;;;;;;;;;:,:;;;;;;;;;:      ;;;;;;;;;;:,;;;;;;;;;;   
    `;;;;;;;;;;;;;;;;;;;.        ;;;;;;;;;;;;;;;;;;;;    
      ;;;;;;;;;;;;;;;;;           :;;;;;;;;;;;;;;;;:     
       ,;;;;;;;;;;;;;,              ;;;;;;;;;;;;;;       
         .;;;;;;;;;`                  ,;;;;;;;;:         
                                                         
                                                         
                                                         
                                                         
    ;;;   ;;;;;`  ;;;;:  .;;  ;; ,;;;;;, ;;. `;,  ;;;;   
    ;;;   ;;:;;;  ;;;;;; .;;  ;; ,;;;;;: ;;; `;, ;;;:;;  
   ,;:;   ;;  ;;  ;;  ;; .;;  ;;   ,;,   ;;;,`;, ;;  ;;  
   ;; ;:  ;;  ;;  ;;  ;; .;;  ;;   ,;,   ;;;;`;, ;;  ;;. 
   ;: ;;  ;;;;;:  ;;  ;; .;;  ;;   ,;,   ;;`;;;, ;;  ;;` 
  ,;;;;;  ;;`;;   ;;  ;; .;;  ;;   ,;,   ;; ;;;, ;;  ;;  
  ;;  ,;, ;; .;;  ;;;;;:  ;;;;;: ,;;;;;: ;;  ;;, ;;;;;;  
  ;;   ;; ;;  ;;` ;;;;.   `;;;:  ,;;;;;, ;;  ;;,  ;;;;   

0

(EthernetSSLClient)(SSL_WARN)(connected): Socket was dropped unexpectedly (this can be an alternative to closing the connection)

Disconnecting.
Received 2925 bytes in 0.1465 s, rate = 19.97 kbytes/second
```

This library relies on the functions and features of the underlying Ethernet libraries and can't provide any magics if the Ethernet libraries are not working.

I suggest you 

1. Test by using the NativeEthernet library directly, not using this [EthernetWebServer_SSL library](https://github.com/khoih-prog/EthernetWebServer_SSL) to see if the SSL layer is working OK there
2. If not, raise an issue on the NativeEthernet library so that the issue can be solved correctly
3. If OK, update here so that I can spend some time to figure out why.

Good Luck,

closed 2021-06-04 by khoih-prog

khoih-prog, 2021-09-04:
Hi @gloveboxes 

Could you please try again using [**QNEthernet** library](https://github.com/ssilverman/QNEthernet) ?

In case you haven't tried it, have a quick look at this very good [**New lwIP-based Ethernet library for Teensy 4.1**](https://forum.pjrc.com/threads/68066-New-lwIP-based-Ethernet-library-for-Teensy-4-1)

---

### Major Releases v1.6.0

1. Add support to [QNEthernet Library](https://github.com/ssilverman/QNEthernet) for Teensy 4.1 built-in Ethernet
2. Update examples with new features