khoih-prog/EthernetWebServer_SSL #8: Library don't work well in Adafruit Grand Central M4

bluestarhcj27, 2021-06-05:
Hello.
I am trying to use this library in my test but I don't know well why it doesn't work.
I am testing the Ethernet client SSL in the Adafruit Grand Central with ATECC608 and Ethernet Shield(https://www.mikroe.com/eth-wiz-click).
[image]
This image is the connection state between Adafruit and Ethernet Shield.
I knew that I should use the ethernet2 to use this shield.( define.h file   Line 288 : #define USE_ETHERNET2   true )
Terminal Result is the following.

Start WebClient_SSL on SAMD51 ADAFRUIT_GRAND_CENTRAL_M4 with W5x00 using Ethernet2 Library
EthernetWebServer_SSL v1.3.1
[ETHERNET_WEBSERVER] =========== USE_ETHERNET2 ===========
[ETHERNET_WEBSERVER] Default SPI pinout:
[ETHERNET_WEBSERVER] MOSI: 66
[ETHERNET_WEBSERVER] MISO: 64
[ETHERNET_WEBSERVER] SCK: 65
[ETHERNET_WEBSERVER] SS: 53
[ETHERNET_WEBSERVER] =========================
[ETHERNET_WEBSERVER] Boardh : SAMD51 ADAFRUIT_GRAND_CENTRAL_M4 , setCsPin: 10

This is the first time to use this library, so I don't know well which I have mistaken.
Please give advice.
Thanks in advance!

khoih-prog, 2021-06-06:
Hi

Thanks you your interests in the library.

From the terminal output, you can know that **your CS pin is 53 (`SS: 53`) on SAMD51 ADAFRUIT_GRAND_CENTRAL_M4.**

Because there are too many boards/shields on the market, the default SS is pre-selected **most popular usage of SS=10**

Therefore, you have to add the following line in [**WebClient_SSL defines.h**#L93-L94](https://github.com/khoih-prog/EthernetWebServer_SSL/blob/main/examples/WebClient_SSL/defines.h#L93-L95) from

```
#elif defined(ADAFRUIT_GRAND_CENTRAL_M4)    
  #define BOARD_TYPE      "SAMD51 ADAFRUIT_GRAND_CENTRAL_M4"
#elif defined(ADAFRUIT_FEATHER_M4_EXPRESS)
```

to 

```
#elif defined(ADAFRUIT_GRAND_CENTRAL_M4)    
  #define BOARD_TYPE      "SAMD51 ADAFRUIT_GRAND_CENTRAL_M4"
  #undef USE_THIS_SS_PIN 
  #define USE_THIS_SS_PIN       53
#elif defined(ADAFRUIT_FEATHER_M4_EXPRESS)
```

Certainly you have to connect `ADAFRUIT_GRAND_CENTRAL_M4` to `ETH-WIZ-CLICK` as follows:

ADAFRUIT_GRAND_CENTRAL_M4    <->   ETH-WIZ-CLICK`

MOSI(66)  <->  SDI (MOSI)
MiSO(64)  <->  SDO (MiSO)
SCK(65)    <-> SCK
SS(53)       <-> CS

You now can know how useful the terminal output to print these lines

```
[ETHERNET_WEBSERVER] Default SPI pinout:
[ETHERNET_WEBSERVER] MOSI: 66
[ETHERNET_WEBSERVER] MISO: 64
[ETHERNET_WEBSERVER] SCK: 65
[ETHERNET_WEBSERVER] SS: 53
[ETHERNET_WEBSERVER] Board : SAMD51 ADAFRUIT_GRAND_CENTRAL_M4 , setCsPin: 10
```

closed 2021-06-06 by khoih-prog

khoih-prog, 2021-06-06:
Just inform you that the SPI bus is on ICSP connector

[image]

bluestarhcj27, 2021-06-06:
Thanks for your quick response.
I saw the Adafruit Grand Central PCB board in (https://github.com/adafruit/Adafruit-Grand-Central-PCB).
And I found the MISO, MOSI, SCK pin in the connector like in the following images.
[image]
[image]
[image]
[image]
[image]

Through these, I notice that the pin number of the MISO is 48 and MOSI is 49, SCK is 50.
But you mentioned like following.
MOSI(66) <-> SDI (MOSI)
MiSO(64) <-> SDO (MiSO)
SCK(65) <-> SCK
SS(53) <-> CS

How is it analyzed?

khoih-prog, 2021-06-06:
Just connect as follows

MOSI(66) == MOSI(ICSP 4) <-> SDI (MOSI) of `ETH-WIZ-CLICK`
MISO(64) == MiSO(ICSP 1) <-> SDO (MiSO) of `ETH-WIZ-CLICK`
SCK(65)  == SCK(ICSP 3) <-> SCK of `ETH-WIZ-CLICK`
SS(53) <-> CS of `ETH-WIZ-CLICK`

On the board, pin MOSI(66) connected to MOSI(ICSP 4) , etc.

bluestarhcj27, 2021-06-12:
I tried to join the connections as you advise.
But the result is like the following.

Start WebClient_SSL on SAMD51 ADAFRUIT_GRAND_CENTRAL_M4 with W5x00 using Ethernet2 Library
EthernetWebServer_SSL v1.3.1
[ETHERNET_WEBSERVER] =========== USE_ETHERNET2 ===========
[ETHERNET_WEBSERVER] Default SPI pinout:
[ETHERNET_WEBSERVER] MOSI: 66
[ETHERNET_WEBSERVER] MISO: 64
[ETHERNET_WEBSERVER] SCK: 65
[ETHERNET_WEBSERVER] SS: 53
[ETHERNET_WEBSERVER] =========================
[ETHERNET_WEBSERVER] Board : SAMD51 ADAFRUIT_GRAND_CENTRAL_M4 , setCsPin: 53

Let me explain the ATECC608 connection state by image
[image]
[image]

So I tried 2 kinds of tests.
1. the WebClient example from this library and the result is good like the following.

@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@

Starting WebClient on SAMD51 ADAFRUIT_GRAND_CENTRAL_M4 with W5x00 using Ethernet Library
EthernetWebServer_SSL v1.3.1
[ETHERNET_WEBSERVER] =========== USE_ETHERNET ===========
[ETHERNET_WEBSERVER] Default SPI pinout:
[ETHERNET_WEBSERVER] MOSI: 66
[ETHERNET_WEBSERVER] MISO: 64
[ETHERNET_WEBSERVER] SCK: 65
[ETHERNET_WEBSERVER] SS: 53
[ETHERNET_WEBSERVER] =========================
[ETHERNET_WEBSERVER] Board : SAMD51 ADAFRUIT_GRAND_CENTRAL_M4 , setCsPin: 53
=========================
Currently Used SPI pinout:
MOSI:66
MISO:64
SCK:65
SS:53
=========================
Using mac index = 12
Connected! IP address: 192.168.0.165

Starting connection to server...
Connected to server
HTTP/1.1 200 OK
Server: nginx/1.4.2
Date: Sat, 12 Jun 2021 19:42:00 GMT
Content-Type: text/plain
Content-Length: 2263
Last-Modified: Wed, 02 Oct 2013 13:46:47 GMT
Connection: close
Vary: Accept-Encoding
ETag: "524c23c7-8d7"
Accept-Ranges: bytes

           `:;;;,`                      .:;;:.           
        .;;;;;;;;;;;`                :;;;;;;;;;;:     TM 
      `;;;;;;;;;;;;;;;`            :;;;;;;;;;;;;;;;      
     :;;;;;;;;;;;;;;;;;;         `;;;;;;;;;;;;;;;;;;     
    ;;;;;;;;;;;;;;;;;;;;;       .;;;;;;;;;;;;;;;;;;;;    
   ;;;;;;;;:`   `;;;;;;;;;     ,;;;;;;;;.`   .;;;;;;;;   
  .;;;;;;,         :;;;;;;;   .;;;;;;;          ;;;;;;;  
  ;;;;;;             ;;;;;;;  ;;;;;;,            ;;;;;;. 
 ,;;;;;               ;;;;;;.;;;;;;`              ;;;;;; 
 ;;;;;.                ;;;;;;;;;;;`      ```       ;;;;;`
 ;;;;;                  ;;;;;;;;;,       ;;;       .;;;;;
`;;;;:                  `;;;;;;;;        ;;;        ;;;;;
,;;;;`    `,,,,,,,,      ;;;;;;;      .,,;;;,,,     ;;;;;
:;;;;`    .;;;;;;;;       ;;;;;,      :;;;;;;;;     ;;;;;
:;;;;`    .;;;;;;;;      `;;;;;;      :;;;;;;;;     ;;;;;
.;;;;.                   ;;;;;;;.        ;;;        ;;;;;
 ;;;;;                  ;;;;;;;;;        ;;;        ;;;;;
 ;;;;;                 .;;;;;;;;;;       ;;;       ;;;;;,
 ;;;;;;               `;;;;;;;;;;;;                ;;;;; 
 `;;;;;,             .;;;;;; ;;;;;;;              ;;;;;; 
  ;;;;;;:           :;;;;;;.  ;;;;;;;            ;;;;;;  
   ;;;;;;;`       .;;;;;;;,    ;;;;;;;;        ;;;;;;;:  
    ;;;;;;;;;:,:;;;;;;;;;:      ;;;;;;;;;;:,;;;;;;;;;;   
    `;;;;;;;;;;;;;;;;;;;.        ;;;;;;;;;;;;;;;;;;;;    
      ;;;;;;;;;;;;;;;;;           :;;;;;;;;;;;;;;;;:     
       ,;;;;;;;;;;;;;,              ;;;;;;;;;;;;;;       
         .;;;;;;;;;`                  ,;;;;;;;;:         
                                                         
                                                         
                                                         
                                                         
    ;;;   ;;;;;`  ;;;;:  .;;  ;; ,;;;;;, ;;. `;,  ;;;;   
    ;;;   ;;:;;;  ;;;;;; .;;  ;; ,;;;;;: ;;; `;, ;;;:;;  
   ,;:;   ;;  ;;  ;;  ;; .;;  ;;   ,;,   ;;;,`;, ;;  ;;  
   ;; ;:  ;;  ;;  ;;  ;; .;;  ;;   ,;,   ;;;;`;, ;;  ;;. 
   ;: ;;  ;;;;;:  ;;  ;; .;;  ;;   ,;,   ;;`;;;, ;;  ;;` 
  ,;;;;;  ;;`;;   ;;  ;; .;;  ;;   ,;,   ;; ;;;, ;;  ;;  
  ;;  ,;, ;; .;;  ;;;;;:  ;;;;;: ,;;;;;: ;;  ;;, ;;;;;;  
  ;;   ;; ;;  ;;` ;;;;.   `;;;:  ,;;;;;, ;;  ;;,  ;;;;   

Disconnecting from server...

@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@

2. I test the ATECC608 dev board with the example of the ECCX08RandomNumber in the ArduinoECCX08 library and the result is good.

@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@
ECCX08 Serial Number = 0123DD06967409B4EE

Hi there, in order to generate a new self signed cert for your board, we'll need the following information ...

Please enter the issue year of the certificate? (2000 - 2031) [2019]: 2021
Please enter the issue month of the certificate? (1 - 12) [1]: 6
Please enter the issue day of the certificate? (1 - 31) [1]: 13
Please enter the issue hour of the certificate? (0 - 23) [0]: 9
Please enter how many years the certificate is valid for? (1 - 31) [31]: 2
What slot would you like to use for the private key? (0 - 4) [0]: 1
What slot would you like to use for storage? (8 - 15) [8]: 9
Would you like to generate a new private key? (Y/n) [Y]: y

Here's your self signed cert, enjoy!

-----BEGIN CERTIFICATE-----
MIIBKjCB0aADAgECAgEBMAoGCCqGSM49BAMCMB0xGzAZBgNVBAMTEjAxMjNERDA2OTY3NDA5QjRF
RTAeFw0yMTA2MTMwOTAwMDBaFw0yMzA2MTMwOTAwMDBaMB0xGzAZBgNVBAMTEjAxMjNERDA2OTY3
NDA5QjRFRTBZMBMGByqGSM49AgEGCCqGSM49AwEHA0IABPLo9OF91E39kFuOEn8VQ1oLqoyLydIl
Gmx1zAc0BLWqxduMTUtObeIX+1cUhB+zQIxRS1RdAxZm6RMQumZ/Nk6jAjAAMAoGCCqGSM49BAMC
A0gAMEUCIQDDkzIm8EVPU6CXbzS74QC9ljoyODHo3vFfZASLrmZLQwIgdZFb0dd2gh3ipXPTGCkw
OsI2JOQRWbwkX3/f/Z65a28=
-----END CERTIFICATE-----

SHA1: 2ba2664b95678420559cbd667f76e0530ea3665d

@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@

So I analyzed the firmware deeply(library deepth) and find in the Dhcp.cpp / int DhcpClass::request_DHCP_lease(),
don't pass the if(_dhcp_state == STATE_DHCP_DISCOVER)               : line74

I would like to know the reason for this error.
Thank you in advance

khoih-prog, 2021-06-12:
Good news that you're moving in good direction.

I'm sorry I can't help you any further as I don't have the exact same hardware to test.

I just can suggest you use the other boards, tested OK in the README, follow the steps there to be OK, then use the exact same steps in your case.

You can also see that `Dhcp.cpp` belongs to one of the Ethernet (Ethernet, Ethenet2, Ethernet3, EthernetLarge, etc) libraries, and have nothing to do with this [**EthernetWebServer_SSL** library](https://github.com/khoih-prog/EthernetWebServer_SSL). Simplify your code to use Ethernet only library, then post an issue in the corresponding library.

If not working, ask for help in the Adafruit or Arduino forum.

Good Luck,

bluestarhcj27, 2021-06-13:
Your analysis was helpful.
It works well.
Thank you for your advice!
The result is the following.

Start WebClient_SSL on SAMD51 ADAFRUIT_GRAND_CENTRAL_M4 with W5x00 using Ethernet Library
EthernetWebServer_SSL v1.3.1
[ETHERNET_WEBSERVER] =========== USE_ETHERNET ===========
[ETHERNET_WEBSERVER] Default SPI pinout:
[ETHERNET_WEBSERVER] MOSI: 66
[ETHERNET_WEBSERVER] MISO: 64
[ETHERNET_WEBSERVER] SCK: 65
[ETHERNET_WEBSERVER] SS: 53
[ETHERNET_WEBSERVER] =========================
[ETHERNET_WEBSERVER] Board : SAMD51 ADAFRUIT_GRAND_CENTRAL_M4 , setCsPin: 53
=========================
Currently Used SPI pinout:
MOSI:66
MISO:64
SCK:65
SS:53
=========================
Using mac index = 2
Connected! IP address: 192.168.0.166
Connecting to : www.arduino.cc, port : 443
Connected to 104.18.29.45
Took: 955
Using micros()
2 74 C8 F1 6D F0 48 66 9C E0 6E 16 C4 D5 E1 3C 
Using analogRead()
47 56 5C 62 63 64 65 64 62 63 65 65 70 69 67 65 
HTTP/1.1 200 OK
Date: Sun, 13 Jun 2021 11:55:39 GMT
Content-Type: text/plain
Transfer-Encoding: chunked
Connection: close
Last-Modified: Wed, 02 Oct 2013 13:46:47 GMT
Vary: Accept-Encoding
Strict-Transport-Security: max-age=500; includeSubDomains
X-Content-Type-Options: nosniff
X-XSS-Protection: 1; mode=block
CF-Cache-Status: DYNAMIC
cf-request-id: 0aa6d34e9100000b69dd1b4000000001
Expect-CT: max-age=604800, report-uri="https://report-uri.cloudflare.com/cdn-cgi/beacon/expect-ct"
Server: cloudflare
CF-RAY: 65eb2190e82d0b69-OSL
alt-svc: h3-27=":443"; ma=86400, h3-28=":443"; ma=86400, h3-29=":443"; ma=86400, h3=":443"; ma=86400

8d7

           `:;;;,`                      .:;;:.           
        .;;;;;;;;;;;`                :;;;;;;;;;;:     TM 
      `;;;;;;;;;;;;;;;`            :;;;;;;;;;;;;;;;      
     :;;;;;;;;;;;;;;;;;;         `;;;;;;;;;;;;;;;;;;     
    ;;;;;;;;;;;;;;;;;;;;;       .;;;;;;;;;;;;;;;;;;;;    
   ;;;;;;;;:`   `;;;;;;;;;     ,;;;;;;;;.`   .;;;;;;;;   
  .;;;;;;,         :;;;;;;;   .;;;;;;;          ;;;;;;;  
  ;;;;;;             ;;;;;;;  ;;;;;;,            ;;;;;;. 
 ,;;;;;               ;;;;;;.;;;;;;`              ;;;;;; 
 ;;;;;.                ;;;;;;;;;;;`      ```       ;;;;;`
 ;;;;;                  ;;;;;;;;;,       ;;;       .;;;;;
`;;;;:                  `;;;;;;;;        ;;;        ;;;;;
,;;;;`    `,,,,,,,,      ;;;;;;;      .,,;;;,,,     ;;;;;
:;;;;`    .;;;;;;;;       ;;;;;,      :;;;;;;;;     ;;;;;
:;;;;`    .;;;;;;;;      `;;;;;;      :;;;;;;;;     ;;;;;
.;;;;.                   ;;;;;;;.        ;;;        ;;;;;
 ;;;;;                  ;;;;;;;;;        ;;;        ;;;;;
 ;;;;;                 .;;;;;;;;;;       ;;;       ;;;;;,
 ;;;;;;               `;;;;;;;;;;;;                ;;;;; 
 `;;;;;,             .;;;;;; ;;;;;;;              ;;;;;; 
  ;;;;;;:           :;;;;;;.  ;;;;;;;            ;;;;;;  
   ;;;;;;;`       .;;;;;;;,    ;;;;;;;;        ;;;;;;;:  
    ;;;;;;;;;:,:;;;;;;;;;:      ;;;;;;;;;;:,;;;;;;;;;;   
    `;;;;;;;;;;;;;;;;;;;.        ;;;;;;;;;;;;;;;;;;;;    
      ;;;;;;;;;;;;;;;;;           :;;;;;;;;;;;;;;;;:     
       ,;;;;;;;;;;;;;,              ;;;;;;;;;;;;;;       
         .;;;;;;;;;`                  ,;;;;;;;;:         
                                                         
                                                         
                                                         
                                                         
    ;;;   ;;;;;`  ;;;;:  .;;  ;; ,;;;;;, ;;. `;,  ;;;;   
    ;;;   ;;:;;;  ;;;;;; .;;  ;; ,;;;;;: ;;; `;, ;;;:;;  
   ,;:;   ;;  ;;  ;;  ;; .;;  ;;   ,;,   ;;;,`;, ;;  ;;  
   ;; ;:  ;;  ;;  ;;  ;; .;;  ;;   ,;,   ;;;;`;, ;;  ;;. 
   ;: ;;  ;;;;;:  ;;  ;; .;;  ;;   ,;,   ;;`;;;, ;;  ;;` 
  ,;;;;;  ;;`;;   ;;  ;; .;;  ;;   ,;,   ;; ;;;, ;;  ;;  
  ;;  ,;, ;; .;;  ;;;;;:  ;;;;;: ,;;;;;: ;;  ;;, ;;;;;;  
  ;;   ;; ;;  ;;` ;;;;.   `;;;:  ,;;;;;, ;;  ;;,  ;;;;   

0

(EthernetSSLClient)(SSL_WARN)(connected): Socket was dropped unexpectedly (this can be an alternative to closing the connection)

Disconnecting.
Received 2925 bytes in 0.4417 s, rate = 6.62 kbytes/second

bluestarhcj27, 2021-06-14:
Hi. Could you give me an answer to what I am not sure as I mentioned above?

khoih-prog, 2021-06-14:
You have to read / research more yourself and don't rely too much on other people. Doing this way, it'll be beneficial for you in the long run.

Just a simple Google search can help you.

For example, [HTTP/HTTPS responses](https://web.stanford.edu/~ouster/cgi-bin/cs142-fall10/lecture.php?topic=http)

```
HTTP/1.1 200 OK
Date: Thu, 24 Jul 2008 17:36:27 GMT
Server: Apache-Coyote/1.1
Content-Type: text/html;charset=UTF-8
Content-Length: 1846

<html>
...
</html>
```

For HTTPS, **requests and responses are identical to HTTP**, except request and response messages are **transmitted using SSL (Secure Sockets Layer) or its successor TLS (Transport Layer Security).**

Please don't post more on the issue unless you prove there is a bug in the library. I have more interesting and better work to do.

Even the issue title is not correct :

**`Library don't work well in Adafruit Grand Central M4 #8`**

while you don't know how to do correctly. Hastily blaming the library proving you're careless and abusing other people's time.

bluestarhcj27, 2021-06-14:
Hi.
Sorry for delaying your valuable work.
As you mentioned the title which I set before was not reasonable.
So I changed it reasonably.
As you know, finally I knew that I should add the pin setting additionally according to the board with your help and knew that there is no error in the library.
Thank you for your effort in creating a good library.