khoih-prog/EthernetWebServer_SSL #14: error: 'ip6_addr_t' does not name a type

mikeme2009, 2022-02-23:
Visual Studio Code v1.64.2
Esp32-gateway-rev-f
OS: Windows 10

Context:
I made a project that use esp32, mqtt, w5500 and I need your library to get esp32 config from my server (SSL).
I installed the khoih-prog / EthernetWebServer_SSL library @ 1.7.4 and when I compiled I got the errors:
C:/Users/xxxxxxx/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/tcpip_adapter/tcpip_adapter.h:62:5: error: 'ip6_addr_t' does not name a type
     ip6_addr_t ip; /**< Interface IPV6 address */
C:/Users/xxxxxx/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/tcpip_adapter/tcpip_adapter.h:407:72: error: 'ip6_addr_t' has not been declared
 esp_err_t tcpip_adapter_get_ip6_linklocal(tcpip_adapter_if_t tcpip_if, ip6_addr_t *if_ip6);

if I uninstall  EthernetWebServer_SSL library, Visual Studio Code will compile without any error.

khoih-prog, 2022-02-23:
I suggest you try to compile your project using Arduino IDE v1.8.19 first to see if there any issue with your code.

Then move on to more knowledge-demanding PlatformIO. Also be careful PIO doesn't support latest ESP32 core v2.0.2 which most libraries are relying on. Check some issues there in PIO forum about how to install latest ESP32 core.

I won't spend time with your PIO support and have have to post and ask for help there.

I'm closing the issue now, and won't reopen until you

1. Use Arduino IDE and have compile error
2. Post the [**Minimal, Reproducible Example**](https://stackoverflow.com/help/minimal-reproducible-example) so that anyone can duplicate the issue without wasting time.

Good Luck,

khoih-prog, 2022-02-23:
I also suggest you use the better ESP32-based WT32_ETH01, which have the built-in and transparent Ethernet / SSL support of ESP32 core.

[image]

closed 2022-02-23 by khoih-prog