khoih-prog/EthernetWebServer_SSL #16: Licensing of SSLClient

celeron55, 2022-03-18:
It appears this project is using OPEnSLab-OSU/SSLClient, which is a GPL 3.0 licensed library.

However, in this project's code base the library has been marked as if it was MIT licensed.

What happened? Is there permission to do this?

khoih-prog, 2022-03-18:
Thanks for the info, which I wasn't aware of.

Will modify the licence ASAP to `GPLv3.0`

khoih-prog, 2022-03-19:
Done with [**EthernetWebServer_SSL** releases v1.7.7](https://github.com/khoih-prog/EthernetWebServer_SSL/releases/tag/v1.7.7). Thanks.

---

### Releases v1.7.7

1. Change license from `MIT` to `GPLv3`. Check [**Licensing of SSLClient** #16](https://github.com/khoih-prog/EthernetWebServer_SSL/issues/16)

closed 2022-03-19 by khoih-prog