khoih-prog/EthernetWebServer_SSL #24: Question about aws iot example

simogaspa84, 2022-09-06:
Hi Mr @khoih-prog  .. 

I was following your example here ... 

https://github.com/khoih-prog/EthernetWebServer_SSL/blob/main/examples/AWS_IoT/AWS_IoT.ino

I have a question about this variable  

`SSLClientParameters mTLS = SSLClientParameters::fromPEM(my_cert, sizeof my_cert, my_key, sizeof my_key);`

I don't find where you use or where should be called.. 
I think it is missing in the example because it is mndatory for aws connection .. 

Please let me know .. 

Thanks for your work

khoih-prog, 2022-09-06:
The `mTLS` use (in function setMutualAuthParams)  is optional to **add a client certificate and enable support for mutual authentication, if needed**. If used, it must be called before EthernetSSLClient starting an SSL connection.

Function prototype

```
void setMutualAuthParams(const SSLClientParameters& params);
```

Used as follows

```
ethClientSSL.setMutualAuthParams(mTLS);
```

You can try to add it before or after this line

https://github.com/khoih-prog/EthernetWebServer_SSL/blob/3bc3b43e9b0da7a3a2f5bb40c679b1721ab6573e/examples/AWS_IoT/AWS_IoT.ino#L353

simogaspa84, 2022-09-06:
Thanks a lot @khoih-prog but for the connection to aws iot core i think it is mandatory for speaking with a device with its own private key and certificate.

khoih-prog, 2022-09-06:
> it is mandatory

I don't know if it's mandatory or not. Possibly new and only for recent AWS. 
Mutual authentication is normally not required for `SSL client`.

Please **try and let us know which way AWS behaves**, as I don't use it quite some time.

closed 2022-09-07 by khoih-prog

simogaspa84, 2022-09-07:
Ok I will try .... 
Do I still need the use of the lib EthernetGeneric.h
if i am using this repo for my ethernet shield ? 

https://github.com/maxgerhardt/EthernetENC.git

Thanks a lot @khoih-prog

khoih-prog, 2022-09-07:
You can use `EthernetENC` library for ENC28J60

It'll be selected if you modify

https://github.com/khoih-prog/EthernetWebServer_SSL/blob/fc79ef0399f898ac6fa0f3897305d52cd6f85db0/examples/AWS_IoT/defines.h#L364-L368

to

```
// Only one if the following to be true
#define USE_ETHERNET_GENERIC  false
#define USE_ETHERNET_ESP8266  false 
#define USE_ETHERNET_ENC      true
#define USE_CUSTOM_ETHERNET   false
```

simogaspa84, 2022-09-07:
It is working 

`[�mum����2-hal-cpu.c:211] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
Joined LAN with IP 192.168.0.46
Attempting MQTT connection...connected
$aws/things/ESP32_TEMP/shadow/update/accepted
$aws/things/ESP32_TEMP/shadow/update/rejected
$aws/things/ESP32_TEMP/shadow/update/delta
$aws/things/ESP32_TEMP/shadow/get/accepted
$aws/things/ESP32_TEMP/shadow/get/rejected
Started updateThing
Published [$aws/things/ESP32_TEMP/shadow/update] {"state": {"reported": {"powerState":"ON"}}}
Done updateThing

`

simogaspa84, 2022-09-08:
With Mutual Authentication